frostsnap/frostsnap #78: Hardware

orangesurf, 2023-07-20:
Interested to know what hardware makes up a BlueBoard / PurpleBoard

musdom, 2023-07-22:
"blue" or "purple" boards are what we internally call the ESP32-C3 devboards we've been using. You can get them from Aliexpress.

- [Purple board](https://www.aliexpress.com/item/1005004341489110.html) Recommended
- [Blue board](https://www.aliexpress.com/item/1005004477968500.html)

The blue board is tricky. You need to get the "NEW ESP32C3" version without USB-Serial chip. You need to get the separate Air101 display hat and solder them together. You might get a different revision of the same LCD display, which requires enabling the r2223 feature flag.

closed 2023-07-22 by musdom

LLFourn, 2023-07-25:
It is likely that we stop supporting these in the near future and move onto different configurations. Hopefully that is clear from the docs?