frostsnap/frostsnap #88: Communicate with a remote coordinator

nickfarrow, 2023-09-08:
Frostsnap (app & cli) should be able to communicate with a remote coordinator.

One coordinator should be "coordinating" and the other coordinators "listen" and pass messages onto signers (maybe after a prompt & Ack).

Currently we have a `UsbSerialManager` which manages some `Serial` connections.

I am thinking we could change this to a `ConnectionManager` which manages some `ConnectionType` connections.

```
pub enum ConnectionType {
    SerialPort,
    HttpPort,
}
```

Which both implement something like
```
pub trait PortConnection {
    fn try_connect(&mut self) -> Result<bool, std::io::Error>;
    fn annouce_self(&mut self) -> Result<(), bincode::error::EncodeError>;
    fn send_message(
        &mut self,
        message: &DeviceReceiveSerial<Downstream>,
    ) -> Result<(), bincode::error::EncodeError>;
    fn try_read_message(
        &mut self,
    ) -> Result<Option<DeviceSendSerial<Downstream>>, bincode::error::DecodeError>;
}
```

The `ConnectionManager` could then easily sync outboxes with remote coordinators somewhere in `poll_read`.

Am I trying to fit too much alongside existing code? `frostsnap_coordinator` is starting to feel a bit complex, I wonder if we can simplify first. Curious for your thoughts on how adding a remote coordinator could most easily be done.

nickfarrow, 2023-09-08:
It would also be good to discuss what communication layer we should use to facilitate this communication.

I was thinking encrypting messages to a public key and broadcasting them on Nostr (with some common channel/hashtag) could be simple? This also means people could swap in their own relay as opposed to relying on a Frostsnap server. And I can defer learning better `client <-> client` networking

LLFourn, 2023-09-18:
I suggest designing this later but just keeping it in mind as we work on single coordinator. For what it's worth I don't think the design would look quite like above. In my mind would be not need for traits or enums. Coordinators are managing a kind of broadcast channel so if we want to do it over the internet we just need a server on the internet to act as a broadcast channel while the coordinators post all the messages to it and read them off. The server can be oblivious to the actual content of the messages.

closed 2025-11-24 by aphanley