frostsnap/frostsnap #110: Display emojis on the device

nickfarrow, 2024-02-05:
Currently characters display as `?`.

Can we allow device labels to be emojis? :house: :office: :mountain_snow: :lobster: :snowflake:.
We can always present devices in lists or dropdowns so the user never needs to type/search them.

LLFourn, 2024-02-05:
Displaying actual text emojis where we just have "❄️" in a string and it shows up as ❄️ on the device is going to be hard but not impossible. I'm not sure if it's desirable though. We could instead make devices have a dedicated icon and then style them separately.

nickfarrow, 2024-02-09:
I was primarily thinking for signing messages that contain emojis or other special characters. I think displaying here is desirable, not sure about device labels.

LLFourn, 2024-02-12:
I propose we start with allowing user to choose a color device icon which gets set up when you set the name.

musdom, 2024-03-05:
Would be awesome but I can't find an embedded-graphics emoji library, plus I'm not sure all the bitmaps would even fit under 4MB. We could probably make display.print support some of the popular emojis by drawing bitmaps when it sees unicode

nickfarrow, 2024-05-15:
I tried to use `U8g2TextStyle::new(fonts::u8g2_font_emoticons21_tr` but I can't see anything even when just writing emoticons..

LLFourn, 2024-05-27:
Some notes to close internal ticket:

convert a .TTF (TrueType Format) into a .BDF (Bitmap Distribution Format) [Overview | Custom Fonts for CircuitPython Displays | Adafruit Learning System](https://learn.adafruit.com/custom-fonts-for-pyportal-circuitpython-display/overview)

[embedded-graphics/bdf: BDF font tools (github.com)](https://github.com/embedded-graphics/bdf)

LLFourn, 2024-06-21:
We could also just send the emoji from the phone whenever a device needed it?

midizzz, 2024-09-24:
would github style random generated avatars work? a few options here https://codegolf.stackexchange.com/questions/82290/generate-a-github-avatar a custom digital snowflake would be 🔥