frostsnap/frostsnap #112: Nonce counter and cache redesign

nickfarrow, 2024-02-08:
* We should separate key generation and nonce generation.
* We should reconsider how we're replenishing nonces during signing
    - Currently the coordinator sends a "replenish index" asking for new nonces to be provided starting from this index.
    - The value of this index should track `NONCE_CACHE_SIZE + nonce_counter`
    - We might need to communicate changes of  `NONCE_CACHE_SIZE` to the coordinator.
* Are there any clever tricks we can do for nonce generation?
* We should consider how we're seeding the nonce rng (prev discussed including pubkey/HMAC from coordinator)

LLFourn, 2024-02-13:
On replenishment I think the right approach is to have the device/signer as the source of truth 
about its nonce counter. Therefore the coordinator shouldn't request nonces starting at a particular index. It can just ask for nonces and the device can tell the coordinator which index they start at.

So each device should have its own `NONCE_CACHE_SIZE` but the coordinator shouldn't need to keep track of this.
When receiving a sign request from a sign request, imagine the device is at nonce index `a` and the request starts at `b` and is of `len`. 

1. The device can infer that the coordinator's last nonce must be at `a + NONCE_CACHE_SIZE` (the device wouldn't give out further than this).
2. It knows its current nonce is at `b + len` and therefore it wants the coordinator to have the nonces in the range `(b + len)..(b + len + NONCE_CACHE_SIZE)`
3. To give it the ones it doesn't have in that range it sends `n_nonces = b + len + NONCE_CACHE_SIZE - (a + NONCE_CACHE_SIZE) = b + len - a` nonces starting at `b + len`

In the actual replenishment protocol piggy backed onto the signing I think the device just sends back `n_nonces`.

nickfarrow, 2024-04-16:
Redesigned in https://github.com/frostsnap/frostsnap/pull/117

closed 2024-04-16 by nickfarrow